theacodes/kicanvas #10: Normal file upload or support any link

smunaut, 2023-04-21:
**Is your feature request related to a problem? Please describe.**
The site https://kicanvas.org/ only lets you drag and drop or post github links.
I have a bunch of stuff not hosted on github but on a private gitea and just pasting links to the "raw" download link to website out of github doesn't seem to do anything.

Other option is drag and drop. I don't have/use any kind of graphical file explorer and all my windows are full screen, I can't _do_ any kind of drag and drop of a local file to a website.

**Describe the solution you'd like**
Ideally both :
 * Support arbitrary link to files to download. You can do a HEAD first to check it's reasonable size to avoid huge file downloads.
 * Add a simple "select file" button for classic upload dialog rather than this drag and drop stuff.

derme302, 2023-04-21:
I can see why this would be difficult, but would at least like to see support for GitLab as well as GitHub

holmesrichards, 2023-07-21:
Also, once a file has been opened, it seems to be impossible to get back to the "drag or github" entry page except by reloading the tab. I'd like an "open" button or menu item on all kicanvas pages.