theacodes/kicanvas #3: "Cannot read properties of undefined (reading 'x')" when loading a schematic

arturo182, 2023-04-21:
Trying to load the bbq20kbd schematic makes KiCanvas sad

Link: https://kicanvas.org/?github=https%3A%2F%2Fgithub.com%2Fsolderparty%2Fbbq20kbd_hw%2Ftree%2Fmain%2F

Console:
[image]

theacodes, 2023-04-25:
(leaving notes for myself since I found the cause but didn't have time to fix it)

This is due to KiCanvas not being able to parse the older Arc format:

```hy
(arc
    (start -0.254 1.016)
    (end -0.254 -1.016)
    (radius (at -0.254 0)
    (length 1.016)
    (angles 90.1 -90.1))
    (stroke (width 0))
    (fill(type none)))
```

We should be able to parse this without an issue, since it's actually easier to deal with than the three-point format they use now.

theacodes, 2023-04-25:
Partially solved by b76bc61a91. That commit fixes this for schematics, but I also need to fix the parser for boards as well, since `gr_arc` and `fp_arc` can appear in the old format as well.

```hy
(gr_arc
    locked
    (start 122.319125 121.852999)
    (end 124.205666 116.094131)
    (angle -18.13822046)
    (layer "F.SilkS")
    (width 0.15)
    (tstamp 0406cd4f-102a-43de-a055-3aa3f08a7ec3))
```

theacodes, 2023-04-25:
Hit a snag with boards where arcs would be oriented the wrong direction. I apparently need to re-do my KiCAD dev environment before I can inspect KiCAD's behavior more closely. With the OHS coming up, I might not get back to this for a week or so.

A workaround is to re-save your files with KiCAD 7, though I understand that's less than ideal.

theacodes, 2023-04-25:
I believe I've fixed this for both boards and schematic files now, though I'm still not really happy with the results. After a lot of frustrating debugging, KiCAD appears to be doing some wonky-ass shit with arc winding and I've been looking at it way too long. I think long-term (as we get closer to KiCAD 8) we can probably just drop support for super old KiCAD 6 files.

commit 724fac93 (2023-04-25): Support parsing legacy arc format in board files

Fixes #3
closed 2023-04-25 by theacodes