theacodes/kicanvas #6: Doesn't work in Safari

dpharris, 2023-04-21:
**Describe the bug**
Window remains blank

**Expected behavior**
Show open to face page.

**Screenshots**
Blank screen

**Environment**

 - OS: OSX  
 - OS version: Ventura M2 laptop
 - Browser:  Safari
 - Browser version:
 - Additional info: 

**Additional context**
Add any other context about the problem here.

gregsadetsky, 2023-06-14:
This also concerns iOS -- nothing is displayed at all on a phone accessing kicanvas.org

theacodes, 2023-06-14:
Safari should be shipping support for that soon, so I'll see where things are when I'm able to come back to this. :)

mengstr, 2023-07-15:
For me it works just fine now.  Running Safari 16.5.1 (18615.2.9.11.7) on Ventura 13.4.1

theacodes, 2023-07-15:
Yeah, looks like this one is fixed. :)

closed 2023-07-15 by theacodes