theacodes/kicanvas #9: No valid KiCAD files found in project

ar1a, 2023-04-21:
Hiya! Trying to get kicanvas to have a look at my (relatively simple? i think) project and it won't show up! Maybe this stems from a mishmash of kicad 6/7,  or maybe something else!

Anyways,
https://github.com/ar1a/sharkpad-rev2

Link that breaks: https://kicanvas.org/?github=https%3A%2F%2Fgithub.com%2Far1a%2Fsharkpad-rev2

ar1a, 2023-04-21:
Sorry, the full error is
```
canvas.js:1619 Error: No valid KiCAD files found in project
    at pR.load_default_file (kicanvas.js:1619:1546)
    at pR.setup_project (kicanvas.js:1619:1131)
    at async HTMLInputElement.<anonymous> (kicanvas.js:1619:861)
```

danielkucera, 2023-04-21:
I have the same:
[image]

danielkucera, 2023-04-21:
uhm, it has to be a direct link to a file, now it works:

[image]

I think it needs some better description.

closed 2023-04-21 by theacodes