theacodes/kicanvas #29: OrangeCrab project loading crashes

yope, 2023-05-25:
URL of project that produces the crash:

https://github.com/orangecrab-fpga/orangecrab-hardware/tree/main/hardware/orangecrab_r0.2

Console logs the following backtrace:

kicanvas-app>  @ 134.486s
12:24:14.787
kicanvas.js:937  Loading project from GitHubFileSystem  @ 134.487s
12:24:14.787
kicanvas.js:937  Loading file OrangeCrab.kicad_pcb  @ 134.487s
12:24:15.723
kicanvas.js:937  Parsing expression with 2679533 chars  @ 135.423s
12:24:15.723
kicanvas.js:1619 Error: Unexpected character while tokenizing atom at index 56: ~
Context: (kicad_pcb (version 20171130) (host pcbnew 5.1.5-52549c5~86~ubuntu18.04.1)
    at tokenize (kicanvas.js:941:144)
    at BV.next (<anonymous>)
    at listify_tokens (kicanvas.js:946:82)
    at vl.next (<anonymous>)
    at Function.from (<anonymous>)
    at listify_tokens (kicanvas.js:946:192)
    at vl.next (<anonymous>)
    at Function.from (<anonymous>)
    at listify_tokens (kicanvas.js:946:192)
    at vl.next (<anonymous>)
12:24:15.724
kicanvas.js:937  Completed  @ 135.424s
﻿

GeorgeIoak, 2023-06-13:
I'm seeing similar crashes with https://github.com/tigard-tools/tigard/tree/master

theacodes, 2023-06-23:
@GeorgeIoak that file was created with KiCAD 5, KiCanvas will not be able to display it.

theacodes, 2023-06-23:
@yope the orangecrab board was also created with KiCAD 5, so it's also not compatible with KiCanvas. I created #34 to track notifying users about incompatibility.

closed 2023-06-23 by theacodes