theacodes/kicanvas #30: Create "Open with" button

beriberikix, 2023-05-28:
A lot of open source projects that are cloud based have a "open with" or "run on" button that through a carefully crafted URL will launch some open source code on a server. Heroku pioneered this idea with the, ["Deploy to Heroku" button](https://devcenter.heroku.com/articles/heroku-button).  You can see more examples at https://shields.io/. I think it would be neat (and good marketing) to have a "Open with KiCanvas" button that anyone could add to say their GitHub readme!

theacodes, 2023-05-28:
wonderful idea!

r4d10n, 2023-08-03:
Thanks for the wonderful project ! I always wished for a way to open Kicad projects in Github !

As a hack solution to this issue, I have created a bookmarklet - Open in KiCanvas:

`javascript:(function(){location.replace('https://kicanvas.org/?github='+location.href)})()`

Now whenever I am in a repo showing Kicad files, I can click the bookmarklet and view the schematic/board files at ease !! 

I guess you can add this bookmarklet on the KiCanvas front page and ask users to drag it to their bookmarklet tab. Ref: https://joeldare.com/creating_a_drag_and_drop_bookmarklet.html

Cheers !