theacodes/kicanvas #33: Footprints containing multiple pads with the same number only render one.

cyber-murmel, 2023-06-22:
> The file you're trying to view. If you don't want the file to be public, let us know and we'll arrange for you to send it privately.
https://kicanvas.org/?github=https://github.com/simplexion/rtl872xd-dev-board/tree/main/source/rtl872xd-dev-board
> A screenshot of the file in KiCAD
will follow up
> A screenshot of the file in KiCanvas
the board page doesn't render anything for me as of now. was still working yesterday...
> A description of the issue
If a footprint has multiple pads with the same number (e.g. ground pads of coaxial connectors) only one of these pads gets filled.

theacodes, 2023-06-22:
Thanks for letting me know!

theacodes, 2023-06-23:
This turned out to be the same issue as #27, so it's been fixed!

closed 2023-06-23 by theacodes