theacodes/kicanvas #36: variables is title block info not parsed in schematic or board view

putyn, 2023-07-12:
same as #17 title block info is displaying variables without parsing them first; issue both in schematic & board; repository with files [here](https://github.com/putyn/epaper/tree/main/epaper-breakout)

kicad
[image]
kicanvas
[image]

theacodes, 2023-07-12:
Fixed for this simple case via c51e67e.

closed 2023-07-12 by theacodes

putyn, 2023-07-13:
that was fast! thanks. 
on last thing, maybe I am nit picking but on schematics with multiple heretical sheets, "Id:" usually displays sheet number/total number of sheets 😄 

example from kicad (4 sheets in total, including root, viewing sheet 2)
[image]

theacodes, 2023-07-13:
Yup, that one is a known issue that requires a little more noodling.