theacodes/kicanvas #37: Can't get it to do anything

holmesrichards, 2023-07-20:
I am running Chrome Version 114.0.5735.198, OS is Linux Mint 21.1 Cinnamon.

If I drag and drop any `.kicad_pro` file onto `https://kicanvas.org/` I see one of two behaviors (I can't reproducibly get one or the other): Either nothing at all happens, or it opens an Open File dialog with the correct file name but pointing into the wrong folder, namely `~/Desktop`.

If I paste `https://github.com/holmesrichards/herovco/blob/main/Hardware/PCB/herovco/herovco.kicad_pro` and hit enter, nothing happens.

holmesrichards, 2023-07-20:
Er, no, still having trouble. I dragged and dropped a folder and had it open before, but now when I try the same thing it opens a new browser tab trying to open the folder as a Chrome tab.

theacodes, 2023-07-21:
I'll have to double check the drop behavior for folders, and also we should be able to figure out the rest of your project files if you paste a URL to a kicad_pro file.

holmesrichards, 2023-07-21:
The behavior seems quite random, sometimes dragging works and sometimes not. It may have to do with previous history of file opening in that tab? I don't know. Right now I can't get it to fail.

commit 8822187a (2023-10-29): Treat GitHub URLs to non-schematic/board blobs as tree links to the containing directory. Closes #37.
closed 2023-10-29 by theacodes