theacodes/kicanvas #38: Themes/colors

holmesrichards, 2023-07-20:
**Is your feature request related to a problem? Please describe.**
I really, thoroughly, utterly dislike the choice of colors used for the schematic display and for the UI in general. I very much prefer the colors used by default in KiCad for the schematic. I very strongly prefer light modes to dark. Blue/green on black is to my taste awful looking and difficult to read.

**Describe the solution you'd like**
Provide some themes functionality or other way to change the colors used.

**Describe alternatives you've considered**

**Additional context**

CapnKernel, 2023-07-21:
Have to say I completely agree.  I'm reluctant to share my designs with others using KiCanvas because the default palette is so distasteful.  And the rest of KiCanvas is so good!

Really hanging out for the option for a light palette.  As the OP said, KiCad's default colours would be fine.

theacodes, 2023-07-21:
> the default palette is so distasteful.

Ouch

> Really hanging out for the option for a light palette. As the OP said, KiCad's default colours would be fine.

Definitely coming before the beta!

holmesrichards, 2023-07-21:
Also, I'm not sure I understand why component leads are drawn in a different color than components, but if you're going to do that then for consistency the "lead" on the ground symbol ought to be different too. As it is, it's a slightly different shade of purple from the wire, which to me looks strange. Of course in KiCad the leads and the components are drawn in the same color, and by default the wire color is entirely different.

[image]

theacodes, 2023-07-21:
Totally understand that folks are going to have different preferences here and I absolutely plan to add themes to KiCanvas before beta, including the default KiCAD scheme. The current KiCanvas theme, [Witch Hazel](https://witchhazel.thea.codes), is designed specifically to help distinguish between differing items on symbols and footprints to ease reviewing.

> Also, I'm not sure I understand why component leads are drawn in a different color than components ... Of course in KiCad the leads and the components are drawn in the same color, and by default the wire color is entirely different.

Since you asked, that's entirely dependent on the color theme, as those are distinct items. Every line in that ground symbol is "body outline" since the pin on that symbol has a length of zero. The cap symbol uses "body outline" for the plates in the symbol, with the pins having their own "pin line" style since those pins aren't zero length.

holmesrichards, 2023-07-21:
Thanks, that makes sense. The outcome, though, still isn't something I like looking at — the straight line of the power symbol juxtaposed to the wire of a very-similar-but-not-the-same color. Zoomed out, in some cases it looks to me like a display glitch. I like it better in KiCad, where the default wire and component colors are entirely different.

jaseg, 2023-07-22:
@theacodes, when you add colorschemes, an approach to this could be to allow for a separate stroke color for power flags that can be different from other components, so users can chose to make power flags the same color as wires.

theacodes, 2023-07-30:
This is done.

closed 2023-07-30 by theacodes