theacodes/kicanvas #40: CD4093 symbol is strange

holmesrichards, 2023-07-21:
https://github.com/holmesrichards/yash/tree/main/Hardware/PCB

KiCad:

[image]

KiCanvas:

[image]

Only half filled, and seems to have AND and OR shapes superimposed.

jaseg, 2023-07-22:
This is KiCad's "demorgan" style feature. I think this feature is a bit disgraced by now, it's not used for new symbols anymore and I guess it will be removed at some point. Currently, a symbol can contain multiple units in each of multiple "demorgan" styles. During rendering, the unit for the selected style has to be super-imposed with special, common "global" units. The 40xx library has a lot of these, but there is a handful of such symbols scattered around other libraries as well.

theacodes, 2023-07-25:
@jaseg thanks for the lead!

Commit 9613b96 adds support for De Morgan body styles, however, there's another bug here in that arcs with fills aren't rendered correctly. I'll have to fix that before I can mark this complete.

[image]

commit 54919ea1 (2023-07-25): Properly fill and stroke polylines, arcs, rectangles, and circles

closes #40
closed 2023-07-25 by theacodes