theacodes/kicanvas #46: Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'background')

vvuk, 2023-10-05:
The version on kicanvas.org, with any file, seems to throw here:

in `kicanvas.js`:
```
    paint() {
        this.document && (this.renderer.background_color = this.theme.background,
```

`this.theme` is null. Tried with both MS Edge and Safari.  (e.g. link from squishy-scsi hw repo: https://kicanvas.org/?github=https%3A%2F%2Fgithub.com%2Fsquishy-scsi%2Fhardware%2Ftree%2Fmain%2Fboards%2Fsquishy )

theacodes, 2023-10-06:
Seems to be working now? https://kicanvas.org/?github=https%3A%2F%2Fgithub.com%2Fsquishy-scsi%2Fhardware%2Ftree%2Fmain%2Fboards%2Fsquishy%2Fmain-unit

I literally just merged a ton of work from a separate branch, so it's possible this got resolved during that work.

closed 2023-10-06 by theacodes

vvuk, 2023-10-06:
Ha :) Yeah, all fixed now. Thanks!