theacodes/kicanvas #50: non-default text & knockout text renders incorrectly

mmalex, 2023-10-24:
hi! I've noticed that knockout text (reverse video) and fancy fonts (georgia on the mac, in this case) don't render correctly.
it looks like the inside/outside rules are wrong.
https://github.com/plinkysynth/scratch_stuff/blob/main/hellokicanvas.kicad_pcb (I tried to make a direct kicanvas link but its not loading?! it loads locally tho)

[image]

[image]

I've also created a more minimal file consisting of the board outline and a single letter 'o' in georgia font, if that's helpful for a simpler repro.
https://github.com/plinkysynth/scratch_stuff/blob/main/minimal.kicad_pcb

sorry if this is a dup, I couldn't see one!

mmalex, 2023-10-24:
(sorry I forgot to say - repros on chrome, firefox and safari mac, all up to date as of today.)
perhaps the fix will be as simple as setting - https://developer.mozilla.org/en-US/docs/Web/SVG/Attribute/fill-rule - to evenodd, rather than nonzero (the default)? but I havent looked at the code.

theacodes, 2023-10-24:
We don't use SVG for rendering. KiCanvas hasn't yet implemented custom
fonts or knockout text, but it's coming..

mmalex, 2023-10-24:
understood! I went and looked after I left my comment and saw that you're tesselating the polygons yourself. awesome, but yeah, makes it tricker. I guess you need a good javascript tesselator with support for self intersecting polygons + odd/even rules? ironically old old gl used to have a one liner more or less to do this (GLUtessellator) but I dont think GLU made it into the webgl age :)