theacodes/kicanvas #52: Error: Unexpected character at index 16286

Weissnix4711, 2023-11-04:
Fails to open [this project](https://github.com/solokeys/solo2-hw). No idea why, but it would be nice to see this working. Other projects seem to open just fine. Tried both browsers I currently have installed. Logs below.

Chromium:

```
Error: Unexpected character at index 16286: ·
Context: 
    (fp_text user · (at 0.275 -0.95 90) (layer F.SilkS)
    at tokenize (kicanvas.js:929:430)
    at Nn.next (<anonymous>)
    at listify_tokens (kicanvas.js:935:82)
    at Os.next (<anonymous>)
    at Function.from (<anonymous>)
    at listify_tokens (kicanvas.js:935:192)
    at Os.next (<anonymous>)
    at Function.from (<anonymous>)
    at listify_tokens (kicanvas.js:935:192)
    at Os.next (<anonymous>)
setup_project @ kicanvas.js:1783
```

LibreWolf:

```
Error: Unexpected character at index 16286: ·
Context: 
    (fp_text user · (at 0.275 -0.95 90) (layer F.SilkS)
    Nn https://kicanvas.org/kicanvas/kicanvas.js:929
    Os https://kicanvas.org/kicanvas/kicanvas.js:935
    Os https://kicanvas.org/kicanvas/kicanvas.js:935
    Os https://kicanvas.org/kicanvas/kicanvas.js:935
    Os https://kicanvas.org/kicanvas/kicanvas.js:935
    Us https://kicanvas.org/kicanvas/kicanvas.js:935
    g https://kicanvas.org/kicanvas/kicanvas.js:935
    Ve https://kicanvas.org/kicanvas/kicanvas.js:936
    #n https://kicanvas.org/kicanvas/kicanvas.js:971
kicanvas.js:1783:978
    setup_project https://kicanvas.org/kicanvas/kicanvas.js:1783
```

theacodes, 2023-11-04:
Is this a KiCAD 5 project?

Weissnix4711, 2023-11-04:
Yes, I believe it is.

Weissnix4711, 2023-11-04:
Ah, I see #34 now. Is kicad 5 support planned/do you have an ETA? Purely out of interest, I'm not trying to pressure you or anything.

theacodes, 2023-11-04:
No, KiCAD 5 support isn't in scope right now.

Weissnix4711, 2023-11-04:
No worries.

How hard do you think it would be to add support? Ie. Should I waste my time trying to learn about KiCAD's file formats, and understand this codebase, to try and add support myself? Or do you think I should not bother?

theacodes, 2023-11-04:
V5 boards are relatively easier, but schematics are a vastly different
format and would be much harder. I wouldn't try to do it at this point in
KiCanvas's development.

theacodes, 2024-10-17:
Closed, duplicate of #34

closed 2024-10-17 by theacodes