theacodes/kicanvas #53: Backslash-escaped characters in text are shown including backspace

hzeller, 2023-11-08:
Text with characters, that are stored backslash-escaped in the file, such as double-quote ( `\"` ) will be shown with the backslash included. Observed with the double-quote character, not sure if there are more.

Here is an text except from the a TRS80 Model100 schematic https://github.com/hzeller/trs80-100-schematic

```scheme
  ...
  (text "White 0.1\" wire bundle\nkeyboard connect" (at 524.51 190.5 90)
    (effects (font (size 1.27 1.27)) (justify left bottom))
    (uuid 689f870a-3693-470a-b767-b22f44e660a0)
  )
 ...
```

The following is the rendering with the text from above (bottom and left of the image) and a in the double-quote in the 4th column and row of the keyboard matrix. The backslash that is used to quote the character in the string is included in the rendering.

KiCAD rendering | KiCanvas rendering
-------------------------|-------------------------------
[image] |[image]

(Sorry, I don't know any TypeScript enough to send a PR)

theacodes, 2023-11-08:
Is this by any chance a KiCAD 5 or KiCAD 6 project?

hzeller, 2023-11-08:
The headline says `(kicad_sch (version 20211123) (generator eeschema)` so I probably compiled KiCAD then, which means it was _almost_ 6.0.

However, I just opened it with 7.0.7 and saved again (now `(kicad_sch (version 20230121) (generator eeschema)`), and the escaping of the strings stayed the same with backslash:

```
$ grep "wire bundle" trs80-100.kicad_sch 
  (text "White 0.1\" wire bundle\nkeyboard connect" (at 524.51 190.5 90)
  (text "Black 0.1\" wire bundle\nkeyboard connect" (at 541.02 254 0)
```

hzeller, 2023-11-15:
For convenience, here the assembled kicanvas link https://kicanvas.org/?github=https%3A%2F%2Fgithub.com%2Fhzeller%2Ftrs80-100-schematic