theacodes/kicanvas #55: Any plan for adding the "Inline source" support ?

XiangYyang, 2023-11-10:
The document [Inline source](https://kicanvas.org/embedding/#inline-source) described a feature called inline source but it has not been implemented yet.

Do you have any plans to add this feature?

It's useful for custom rendering such as Gitea. We can echo the sources into HTML and import kicanvas.js, and the file will be rendered.

## A possible implementation

I found a possible implementation for this feature. The HTML tag was processed in the file `src/kicanvas/elements/kicanvas-embed.ts`. And all `src` attributes will be appended to a list. Then, `FetchFileSystem` will load them. Finally, they were converted to `File` objects. 

So a simple way is changing the `KiCanvasEmbedElement.load_src` and `FetchFileSystem` in `vfs.ts` to implement the creation of File objects from the inline text. 

I'm trying implementing it. If that is possible, I will create a pull request later.

theacodes, 2023-11-10:
Yes, I do plan on implementing it. There are no timelines. If you'd like to
try it, go for it.

eeyrjmr, 2024-12-31:
> It's useful for custom rendering such as Gitea. We can echo the sources into HTML and import kicanvas.js, and the file will be rendered.

This is my usecase as well as I have my personal project manged via gitea so being able to view inline would be <3

shabh2412, 2025-03-19:
Looking forward to this feature. 🙌🏼 @theacodes