theacodes/kicanvas #73: Load ok from file but not from github link

richClubb, 2024-01-11:
Hello. Thanks for this tool, it's great!

I've found a weird one:
* Kicanvas is happy to load `https://github.com/richClubb/wifi_temp_sensor/tree/main/hardware` and gives me the board and the schematic. It can also load the schematic and PCB from that directory no problem
* It wont load `https://github.com/richClubb/wifi_temp_sensor/tree/dev/v0_3/hardware` or `https://github.com/richClubb/wifi_temp_sensor/tree/dev/v1_0/hardware` but will load the schematic and pcb files in those directories with no problem. `https://github.com/richClubb/wifi_temp_sensor/blob/dev/v1_0/hardware/wifi_temp_sensor.kicad_sch` and `https://github.com/richClubb/wifi_temp_sensor/blob/dev/v1_0/hardware/wifi_temp_sensor.kicad_pcb` for example

It's not critical, just very weird.

richClubb, 2024-01-11:
I'll try and leave the 2 non working branches alone for a bit and just branch off to do any more dev.

theacodes, 2024-01-11:
it's okay! I can always refer to the specific commit when testing.

XiangYyang, 2025-12-16:
Thanks for letting us know it !!. It looks like a bug in `Github.parse_url`

https://github.com/theacodes/kicanvas/blob/b9b68cb997f2c8274c8b2e6ba454c627a6634a64/src/kicanvas/services/github.ts#L52-L62

The function use `https://api.github.com/repos/richClubb/wifi_temp_sensor/contents/v1_0/hardware?ref=dev/v1_0` instead of `https://api.github.com/repos/richClubb/wifi_temp_sensor/contents/hardware?ref=dev/v1_0`