theacodes/kicanvas #93: Should kicanvas-embed work locally?

higaski, 2024-09-14:
I'm almost afraid to ask this question because it seems rather stupid, but I'm really not a "Web" person so please forgive me.

Should embedding work locally when opening a simple html file with a browser?
Let's assume I've a folder containing
```sh
.
├── index.html
├── kicanvas.js
└── simple.kicad_sch
```

with an index.html file looking like this
```html
<!DOCTYPE html>
<html>

<head>
    <script type="module" src="/kicanvas.js"></script>
</head>

<body>

    <p>
        <kicanvas-embed src="/simple.kicad_sch" controls="basic"> </kicanvas-embed>
    </p>

</body>

</html>
```

But sadly this doesn't work. Should it? And if yes, what am I doing wrong?

theacodes, 2024-09-14:
Are you just opening the file directly in your web browser? as in, the URL bar starts with `file://`. If so, I don't think it'll work that way since Chrome has a lot of restrictions on what web pages loaded like that can do.

higaski, 2024-09-14:
Yes

Later on I'd like to embed Kicanvas in some Doxygen generated documentation which supports plain html as well. I simply like to preview my docs locally before pushing them to some webserver. 😄

theacodes, 2024-09-16:
Ah, I don't think there's much I can do to support the use case of opening a file directly like that. Chrome requires a "secure context" (`localhost` or `https`) to access certain APIs. 

I'm going to close this for now, but feel free to reply if you have any other questions.

closed 2024-09-16 by theacodes

higaski, 2024-10-05:
Just in case someone finds this issue, if you need to run it locally just fire up a local HTTP server with python
`python -m http.server --directory <your html folder> --bind 127.0.0.1`