theacodes/kicanvas #101: Unexpected character ~

osresearch, 2024-12-08:
https://kicanvas.org/?github=https%3A%2F%2Fgithub.com%2Ffelixheld%2Fqspimux%2Fblob%2Fmaster%2Fqspimux%2Fqspimux.kicad_pcb

```
Error: Unexpected character at index 32069: ~
Context: 
    (pad ~ smd circle (at 0 0) (size 0.5 0.5) (layers F.Cu F.Mask)
```

It is an older version `(kicad_pcb (version 20171130) (host pcbnew "(5.1.0-rc2-47-g6eb84e42f)")`, so perhaps something has changed in the pcb file format since then.

theacodes, 2024-12-08:
Yeah, the format changed a lot between 5 and 6, so KiCanvas doesn't support 5. We have a bug for showing better error messages for v5 and lower files.

osresearch, 2024-12-08:
I figured it was something like that. I'll keep an eye on #34 then.  Thanks for writing such a useful tool!

closed 2024-12-08 by osresearch

theacodes, 2024-12-08:
Of course!