theacodes/kicanvas #112: KiCad 9 design not loading

matthewlai, 2025-05-20:
https://github.com/matthewlai/GloveElectronics

Thanks!

boluo-O, 2025-05-21:
It seems there was an error when kicanvas parsed Glove.kicad_pcb for embedded base64 file `(embedded_files
		(file
			(name "CONN-SMD_PH2.00-LS-2P.wrl")
			(type model)
			(data |KLUv/aAi ......` . If you just want to temporarily make it display properly, you can try delete this 3D model CONN-SMD_PH2.00-LS-2P.wrl in KiCad or just test this one.

[Glove.kicad_pcb.zip](https://github.com/user-attachments/files/20360424/Glove.kicad_pcb.zip)

matthewlai, 2025-05-22:
Good to know thanks!

andyshinn, 2025-06-29:
I also had to remove any images added as layers on the footprints (look for `(data ` parts in the PCB file).

theacodes, 2025-07-02:
Our parser probably needs to be more robust when encountering unknown/invalid items.

andyshinn, 2025-07-03:
Go ahead and reopen if #117 doesn't seem like it fixes it. It does for me (for embedded PNGs)